kartaview/android #113: Make OSC available on F-Droid

stefannagy, 2017-12-19:
In the [F-Droid](https://f-droid.org/) forum [I asked](https://forum.f-droid.org/t/openstreetcam/1753) if the OSC Android app meets the F-Droid [inclusion criteria](https://f-droid.org/docs/Inclusion_Policy/). I was told
> No. It includes proprietary libraries (com.crashlytics.sdk.android, com.google.android.gms) and custom Maven repositories.

I'd love to see OSC in F-Droid.

ramack, 2017-12-23:
It would be great to have a free (as in free software) variant of OSC, which even respects privacy of the users. So in my eyes it is not only about a second distribution channel, but even if you decide to only publish this great app on play store it makes totally sense to have two build flavors, a free and the "traditional" one, if not even only e free one.

The F-droid inclusion criteria are not just restrictions for developers, they totally make sense in my eyes.

stephan75, 2018-04-22:
About manually install of apk files: they seem to not be recent on https://github.com/openstreetcam/android/releases

PanderMusubi, 2018-10-19:
See also https://github.com/openstreetcam/android/issues/8 and https://gitlab.com/fdroid/rfp/issues/80

tuxayo, 2019-08-19:
Now the bigger problem is that the app is not anymore libre/open source, see #153

PanderMusubi, 2019-08-19:
So we close this issue on closedstreetcam?

tuxayo, 2020-02-20:
Source code of current (v3.0.1) version [released](https://github.com/openstreetcam/android/issues/153#issuecomment-589250886)

tuxayo, 2020-02-25:
Well, nope :(

https://github.com/openstreetcam/android/issues/153#issuecomment-589282074

mnalis, 2023-04-12:
since the issue #153 have been removed, here is last known archive of link above: http://web.archive.org/web/20201107225942/https://github.com/openstreetcam/android/issues/153#issuecomment-589282074