kartaview/android #114: Binaries offered in releases

PanderMusubi, 2018-01-29:
The latest release doesn't offer any binaries. Are these no longer support via releases? If so, remove the links called "Download .apk" in the dialog on http://openstreetview.org wheb clicking on "Get the App". If not, please offer the binaries again for each release.

ec1oud, 2018-08-06:
Is this app supported at all anymore?  It's been half a year since this issue was opened, no response and still not binaries.  Last binary release is from last year.

PanderMusubi, 2018-08-06:
It is indeed awfully quiet and hope someone can show a sign of life.

alexandrui-telenav, 2018-08-06:
Github releases are behind GooglePlay builds, fortunately that doesn't reflect the overall project activity. A colleague will add the next app update to Github too, we're currently testing it before release.

ec1oud, 2018-08-07:
The main thing for me is to solve the login problem (issue #127) so that I can upload the image sequences I already took.  I figured maybe a new release might happen to have that fixed.  But I see the git repo doesn't have very many recent changes either.

ciprianh-telenav, 2018-08-08:
@PanderMusubi you may use the latest apk from 
https://github.com/openstreetcam/android/releases/download/2.3/osc_2.3.0_release.apk
Thank you!

closed 2018-08-08 by ciprianh-telenav

PanderMusubi, 2018-08-09:
Thanks.

Another reason to have Github up to date is that other people can more easily contribute code.