kartaview/android #117: OSC app massively overestimates points when using OBD-II dongle

rich123, 2018-04-27:
The Android OSC app, when using an OBD-II dongle, massively overestimates the points obtained from a sequence in the summary screen and in the points overview prior to uploading the sequence.

Sequence capture summary screen:

[image]

Note that the "estimate" is 100K points.

Sequence screen, pre-uploading:

[image]

Note that the blue circle says "100k" points.

Points breakdown screen:

[image]

Note that the total line says 100K points, but the points values in the table under "Points" do not sum to anywhere near 100K.

I did not notice this effect with plain GPS captures.  It only seems to occur when the app. is also using an OBD-II dongle.

bogdans-telenav, 2018-05-07:
Hi @rich123 thanks for reporting this. This issue was found in a older version, can you please say what version you are using? Our QA will try to reproduce this. If you or the QA confirm that the bug is present in the latest version a bug will be issued. The fix for this bug will most probably not be present in the next release but when we will have a fix for this we will close the bug. Thanks again for the screenshots and detailed bug description.

rich123, 2018-05-07:
Sorry for leaving the version out.

The app reports it is 2.2.0.  I installed it via the play store, so I presume it is the current version.

bogdans-telenav, 2018-05-07:
2.2.0 is the latest version.

ciprianh-telenav, 2018-06-04:
@rich123, the next version (2.3) will bring the fix for this issue. It will be available soon. Thank you for reporting the issue and thank you for your patience.

closed 2018-08-08 by ciprianh-telenav