kartaview/android #120: Cannot login with OSM

tuxayo, 2018-05-22:
- I see the webview
- log in OSM
- authorize the app
- then I'm just back to the login method selection screen like nothing happened

Device: Moto G3 (XT1541 8GB) with LineageOS for microG 14.1 (Android 7.1.2)

Logcat:
```
05-23 00:05:20.360 13277 13277 D LoginManager: login: requested login type osm
05-23 00:05:20.990 13277 13277 I chromium: [INFO:CONSOLE(4)] "Deprecated include of L.Mixin.Events: this property will be removed in future releases, please inherit from L.Evented instead.", source: https://www.openstreetmap.org/assets/application-a3a4f7916baa000a1eba9dd01ca7a62622218f60c2213f983b27a380ce37cdc6.js (4)
05-23 00:05:21.448    97    97 W kswapd0 : type=1400 audit(0.0:787): avc: denied { use } for uid=0 path=2F6465762F6173686D656D202864656C6574656429 dev="tmpfs" ino=1059990 scontext=u:r:kernel:s0 tcontext=u:r:untrusted_app:s0:c512,c768 tclass=fd permissive=0
05-23 00:05:21.448    97    97 W kswapd0 : type=1400 audit(0.0:788): avc: denied { use } for uid=0 path=2F6465762F6173686D656D202864656C6574656429 dev="tmpfs" ino=1059990 scontext=u:r:kernel:s0 tcontext=u:r:untrusted_app:s0:c512,c768 tclass=fd permissive=0
05-23 00:05:27.637 13277 13377 E chromium: [ERROR:gl_context_virtual.cc(39)] Trying to make virtual context current without decoder.
05-23 00:05:27.984 13277 15020 W System.err: java.io.IOException: Unauthorized
05-23 00:05:27.984 13277 15020 W System.err: 	at com.telenav.osv.manager.network.LoginManager$6$1$1.run(LoginManager.java:486)
05-23 00:05:27.984 13277 15020 W System.err: 	at java.lang.Thread.run(Thread.java:761)
```

Is there more info that I can provide?

ciprianh-telenav, 2018-05-23:
Victor, thank you for reporting this. OSM changed some things in the OAuth login. A new version of the app will be available with the fix and some other changes we were working on in the last period. Thank you for your patience.

tuxayo, 2018-05-23:
Great, I feared that it was some specific issue with my case.

tuxayo, 2018-05-26:
And it works with v2.2.1!

Thanks a lot, I'll able to do my first upload, cookie for all! 🍪🍪🍪🍪🍪🍪🍪🍪🍪🍪🍪🍪🍪

closed 2018-05-26 by tuxayo