kartaview/android #121: Unlock autofocus

tuxayo, 2018-05-22:
> You can tap on screen to focus. Long press locks it

> long tap to focus on the area and lock the focus to it

Is there a way to undo this without canceling the recording?
When it's not the best choice to lock the focus then doing a long press by mistake can ruin a recording.

ciprianh-telenav, 2018-05-23:
Victor, you can undo the focus lock by tapping anywhere on the screen, even while recording. If this does not work on your device can you let us which device are you using? Thank you!

tuxayo, 2018-05-23:
I tried before posting the issue and was very confused by the results.
And tried again more thoroughly and still can make it work like it should.

No worries, I'll reopen the issue if I have a problem in a real use case with focus.

closed 2018-05-23 by tuxayo