kartaview/android #122: Crash at launch (SIGSEGV segfault)

tuxayo, 2018-05-22:
I don't know what happened during one of the launches, but it crashed.

[logcat.txt](https://github.com/openstreetcam/android/files/2028824/logcat.txt)

Device: Moto G3 (XT1541 8GB) with LineageOS for microG 14.1 (Android 7.1.2)

tuxayo, 2018-05-22:
Another crash which may or not be related.
[logcat2.txt](https://github.com/openstreetcam/android/files/2028851/logcat2.txt)

Relaunched the app a bunch of time until it recrashed
[logcat3.txt](https://github.com/openstreetcam/android/files/2028879/logcat3.txt)

ec1oud, 2018-08-06:
It crashes every time on my old Galaxy Note too, running KitKat I think.  (Not my main phone anymore, but I tried because of the unable-to-login problem on my newer one.)