kartaview/android #124: Cannot connect to bluetooth ODB2 dongle

rajk1, 2018-07-11:
I have a ScanTool ODBLink MX bluetooth dongle which used to work with the previous version of the OSC Android app. In the new version (2.3.0) it refuses to establish a connection with the same dongle. I can establish connection to the ODBLink MX dongle through other apps such as ODBLink. While trying to make a connection I see only 2 options: BLE and Wi-Fi. There is no option for Bluetooth. Is this a bug?

chrisdebian, 2018-07-11:
Hi, Raj.

I assume that 'BLE' refers to 'Bluetooth Low Energy (BTLE)'. Just to help
others resolve this, did you try this option?

Many thanks,

Chris.

rajk1, 2018-07-11:
Hi Chris,

Yes, I did try the Bluetooth Low Energy option but it refused to connect. It shows the paired Bluetooth devices but when I select the ODB2 dongle it throws an error.

Regards,
Raj

rajk1, 2018-07-11:
Screenshots:
[image]

[image]

alexandrui-telenav, 2018-07-13:
Hi @rajk1, we dropped support for Bluetooth dongles in the last Android version ( BluetoothLE is  supported, just not Bluetooth ).
We had quite some reliability problems with the dongles and the usage wasn't significant so we decided to drop support. 
Unfortunately you were amongst a few who were affected by that. We recommend getting a good Wi-Fi or BLE dongle, there are some samples in the FAQ of the app.

closed 2018-07-13 by alexandrui-telenav

alexandrui-telenav, 2018-07-23:
@rajk1 What I can do is send you a redeem code for a BluetoothLE dongle if you're living in the US.

rajk1, 2018-07-24:
Hi, I live in Canada. I bought the LELink BluetoothLE dongle which is working quite well.