kartaview/android #125: V 2.3.0 of OSC (Android) has lost bluetooth connectivity option

rich123, 2018-07-19:
The OSC update to version 2.3.0 (Android) has lost its previous ability to connect to bluetooth OBD adapters.

The prior version (2.2.0) was working perfectly with this OBD adapter: https://www.scantool.net/obdlink-lxbt/

2.3.0 now only mentions wi-fi and bluetooth le in its new OBD setup screen, and while it sees the bluetooth adapter, it no longer connects to the adapter.

My phone does not have bluetooth le, but does have bluetooth (note that "le" and basic bluetooth are two different protocols).

Several screen shots below:
1) The phone sees the adapter:
[image]

2) OSC 2.3.0 also 'claims' it sees the bluetooth adapter:
[image]

3) But when selecting the adapter, OSC 2.3.0 can no longer access it:
[image]

Please bring back regular "Bluetooth" capability.  Adding "LE" to the mix is fine, but don't drop classic Bluetooth when adding "LE".  Note that 2.2.0 in its alternate setup for OBD devices listed all three of Wi-Fi, Bluetooth, and Bluetooth LE as options.  Please provide the same for 2.3.0.

alexandrui-telenav, 2018-07-23:
Hi @rich123 , we've noticed problems with OBD data coming from Bluetooth dongles and because usage wasn't high we decided to remove support.
I understand that your phone doesn't support BLE, but in case you have other one handy I can send you a redeem code for a BLE dongle instead.

rich123, 2018-07-23:
That is most unfortunate.  I uploaded almost 1000 miles worth of traces, many of brand new roads, mid June with the prior version that supported bluetooth and the bluetooth adapter that I have. 

I would ask you to reconsider that decision.

A redeem code for a BLE adapter will do me no good.  My phone does not have LE, and I have only that phone, so I can't make use of an LE adapter.

And I have tried side loading the older version that supported bluetooth, but it refuses to authenticate using oauth now, and I don't know why (unless you disabled that as well).

rich123, 2018-07-24:
Ok, digging much deeper into my phone specs, it looks like the phone may support BLE.  I have no BLE devices with which to verify this fact however.  So a redeem code may work.

alexandrui-telenav, 2018-07-24:
The main cause for the subpar OBD data were the really cheap Bluetooth dongles that some people got. Not your case, I saw that you have a good one, but with the other dongles providing better data we decided to go ahead with dropping support.

Hope a new dongle makes it better. Where can I send the redeem code? Github doesn't seem to have DM support.

rich123, 2018-07-24:
It is unfortunate you could not simply blacklist the bad adapters instead of completely dropping bluetooth support.

Your question caused me to check my profile, I had not turned on a public email address as of yet.  I have now done so.  You may use the email address that is now visible on my profile here.

rich123, 2018-08-01:
Ok, so I got the BLE adapter via the redeem code.  The phone sees the BLE adapter.  The app will not connect to the (your actual recommended one) BLE adapter.  About half to three quarters of the time, when I select the BLE adapter I get the "Oops, can't find adapter" message.  If I hit "try again" several times or back out and retry several times it eventually gets to the "connecting OBD" dialog where it says "Please wait while we connect to the adapter".

At which point it (the app) remains stuck.  It never connects, it just sits there.  I can cancel, so the app. is not hung or crashed, but it can't connect to the BLE adapter.

I can go back out to the "recording" screen, at which point the blue obd icon will have a tool tip stating "connecting car obd" that never disappears.

So things are still very broken.

gdt, 2018-08-01:
I find it strange that an open-source project would delete working support for a class of devices because some devices in that class are junky.   I agree that having a list of known-broken devices seems quite easy in order to avoid tickets filed by people who have a non-working situation because of that.

alexandrui-telenav, 2018-08-02:
@rich123 Sorry to hear that, we did quite a some testing with that BLE dongle and didn't hear of something similar before either.
Could you please go to Settings -> Report a problem? When you submit that we also get the logs from the past 24h, would help a lot in understanding the problem. Thanks.

rich123, 2018-08-02:
Done.  The description I attached was "Unable to connect to new BLE adapter".

I performed two cycles of trying to connect.  Both included several "unable to find" dialogs, and after several "try agains" it then got to the "connecting to" dialog where things get 'stuck' and do not continue.

Further, I noticed that I had to reset (unplug) the OBD adapter between the two try's to get it to show back up even on the phone's 'available devices' screen.  So it seems like the app is sending some command that is causing the adapter to lock up or otherwise disappear.

alexandrui-telenav, 2018-08-03:
Thanks rich, we received them and will investigate today.

cameliao-telenav, 2018-08-03:
Hi @rich123, thanks for your support and patience. We looked into the issue and tested the case that you described with the same model for the OBD dongle. Apparently the problem is reproducing only on a small range of devices and your device is one of them. Currently, we are working on to fix the issue. We will let you know when is done.
Sorry for the inconvenience, we are trying to fix the problem as soon as possible.

rich123, 2018-08-03:
Ah, so there is some additional issue hiding in the mix.  Ok, good to hear that you are working on a fix.

One clarification, by "device" do you mean the issue is with my phone, or that the issue is with my particular variant of the recommended BLE OBD adapter?

cameliao-telenav, 2018-08-06:
There is an issue in the actual Android system which is reproducible only on some of the mobile phones. Therefore, we need to treat it specifically for that android version.

mvexel, 2019-03-18:
Bump @alexandrui-telenav @cameliao-telenav ..?