kartaview/android #127: unable to login to OSM

ec1oud, 2018-08-06:
The login option is there, I can choose Login with OpenStreetMap, I get a popup dialog with "Authorize access to your account", the checkbox to allow it to read user preferences is checked, I click Grant Access, the progress spinner spins for a while, and... nothing.  I'm apparently still not logged in because I can't upload any tracks, but I'm not getting an error message about what went wrong either.  If I try to upload recordings there is the message at the bottom of the screen, Please log in to upload your recordings.

This is with a Moto Z, Android 8.0.0.  I don't keep Google Play Services up-to-date and never download from the Play Store, to try to avoid some of the spyware.  Got the 2.1.0 release since that's the latest one that you offer on https://github.com/openstreetcam/android/releases

ec1oud, 2018-08-07:
[upload-attempt.log](https://github.com/openstreetcam/android/files/2267510/upload-attempt.log)
Here's some output of adb logcat: a crash from a few days ago, and my last attempt to log in and upload.

ciprianh-telenav, 2018-08-08:
@ec1oud, you may use the latest apk from 
https://github.com/openstreetcam/android/releases/download/2.3/osc_2.3.0_release.apk
It brings the OSM login fix. Thank you!

closed 2018-08-08 by ciprianh-telenav

ec1oud, 2018-08-09:
Yes indeed it works now.  Thanks.