kartaview/android #137: Create/use directories with a more readable timestamp

naoliv, 2018-08-29:
I have directories named `SEQ_1535283100617`, `SEQ_1535293252732`, `SEQ_1535382073557` and `SEQ_1535383428552` and, as you can see, it's not possible to know anything about them (date or time when the sequence was captured, for example)

Could we have some better named directories, possibly using YYYY-MM-DD-HH-MM-SS (or something like this), please?

ciprianh-telenav, 2018-09-07:
Nelson, you should be able to see the date on internal storage/Android/data/com.telenav.streetview/files/osv.

closed 2018-09-07 by ciprianh-telenav

naoliv, 2018-09-07:
There are 2 issues here: if we copy the directories from the phone, we lose the original timestamp from the files.

Also, you will have to agree that it's a PITA when an user makes a lot of trips and will need to find the date/time by taking a look one picture EXIF and, then, manually rename the directory.