kartaview/android #141: Not catching errors while uploading

naoliv, 2018-08-29:
Despite seeing an `Upload finished` message at my screen, OSC doesn't always upload all the files (nor gives any error message about this).
In logcat I can see things like this:

```
08-29 14:16:25.634 24258 24343 D ApiResponseParser: Error when parsing volley error. Volley error: com.android.volley.ServerError
08-29 14:16:25.634 24258 24343 D ApiResponseParser:     at com.android.volley.toolbox.BasicNetwork.performRequest(BasicNetwork.java:191)
08-29 14:16:25.634 24258 24343 D ApiResponseParser:     at com.android.volley.NetworkDispatcher.run(NetworkDispatcher.java:133)
08-29 14:16:25.634 24258 24343 D ApiResponseParser: 
08-29 14:16:25.634 24258 24343 D ApiResponseParser: Parsing error:org.json.JSONException: Value <!DOCTYPE of type java.lang.String cannot be converted to JSONObject
08-29 14:16:25.634 24258 24343 D ApiResponseParser:     at org.json.JSON.typeMismatch(JSON.java:111)
08-29 14:16:25.634 24258 24343 D ApiResponseParser:     at org.json.JSONObject.<init>(JSONObject.java:160)
08-29 14:16:25.634 24258 24343 D ApiResponseParser:     at org.json.JSONObject.<init>(JSONObject.java:173)
08-29 14:16:25.634 24258 24343 D ApiResponseParser:     at com.telenav.osv.manager.network.parser.ApiResponseParser.parse(ApiResponseParser.java:44)
08-29 14:16:25.634 24258 24343 D ApiResponseParser:     at com.telenav.osv.listener.network.OsvRequestResponseListener.onErrorResponse(OsvRequestResponseListener.java:26)
08-29 14:16:25.634 24258 24343 D ApiResponseParser:     at com.android.volley.Request.deliverError(Request.java:549)
08-29 14:16:25.634 24258 24343 D ApiResponseParser:     at com.android.volley.ExecutorDelivery$ResponseDeliveryRunnable.run(ExecutorDelivery.java:104)
08-29 14:16:25.634 24258 24343 D ApiResponseParser:     at android.os.Handler.handleCallback(Handler.java:746)
08-29 14:16:25.634 24258 24343 D ApiResponseParser:     at android.os.Handler.dispatchMessage(Handler.java:95)
08-29 14:16:25.634 24258 24343 D ApiResponseParser:     at android.os.Looper.loop(Looper.java:148)
08-29 14:16:25.634 24258 24343 D ApiResponseParser:     at android.os.HandlerThread.run(HandlerThread.java:61)
08-29 14:16:25.634 24258 24343 W System.err: org.json.JSONException: Value <!DOCTYPE of type java.lang.String cannot be converted to JSONObject
08-29 14:16:25.634 24258 24343 W System.err:    at org.json.JSON.typeMismatch(JSON.java:111)
08-29 14:16:25.634 24258 24343 W System.err:    at org.json.JSONObject.<init>(JSONObject.java:160)
08-29 14:16:25.634 24258 24343 W System.err:    at org.json.JSONObject.<init>(JSONObject.java:173)
08-29 14:16:25.634 24258 24343 W System.err:    at com.telenav.osv.manager.network.parser.ApiResponseParser.parse(ApiResponseParser.java:44)
08-29 14:16:25.634 24258 24343 W System.err:    at com.telenav.osv.listener.network.OsvRequestResponseListener.onErrorResponse(OsvRequestResponseListener.java:26)
08-29 14:16:25.634 24258 24343 W System.err:    at com.android.volley.Request.deliverError(Request.java:549)
08-29 14:16:25.634 24258 24343 W System.err:    at com.android.volley.ExecutorDelivery$ResponseDeliveryRunnable.run(ExecutorDelivery.java:104)
08-29 14:16:25.635 24258 24343 W System.err:    at android.os.Handler.handleCallback(Handler.java:746)
08-29 14:16:25.635 24258 24343 W System.err:    at android.os.Handler.dispatchMessage(Handler.java:95)
08-29 14:16:25.635 24258 24343 W System.err:    at android.os.Looper.loop(Looper.java:148)
08-29 14:16:25.635 24258 24343 W System.err:    at android.os.HandlerThread.run(HandlerThread.java:61)
```

```
08-29 14:16:44.523 24258 24258 E EventBus: Could not dispatch event: class com.telenav.osv.event.network.upload.UploadProgressEvent to subscribing class class com.telenav.osv.ui.fragment.UploadProgressFragment
08-29 14:16:44.523 24258 24258 E EventBus: java.lang.NumberFormatException: Invalid double: "99,1"
08-29 14:16:44.523 24258 24258 E EventBus:      at java.lang.StringToReal.invalidReal(StringToReal.java:63)
08-29 14:16:44.523 24258 24258 E EventBus:      at java.lang.StringToReal.initialParse(StringToReal.java:164)
08-29 14:16:44.523 24258 24258 E EventBus:      at java.lang.StringToReal.parseDouble(StringToReal.java:282)
08-29 14:16:44.523 24258 24258 E EventBus:      at java.lang.Double.parseDouble(Double.java:301)
08-29 14:16:44.523 24258 24258 E EventBus:      at com.telenav.osv.ui.fragment.UploadProgressFragment.updateStats(UploadProgressFragment.java:349)
08-29 14:16:44.523 24258 24258 E EventBus:      at com.telenav.osv.ui.fragment.UploadProgressFragment.onProgressChanged(UploadProgressFragment.java:245)
08-29 14:16:44.523 24258 24258 E EventBus:      at java.lang.reflect.Method.invoke(Native Method)
08-29 14:16:44.523 24258 24258 E EventBus:      at org.greenrobot.eventbus.EventBus.invokeSubscriber(EventBus.java:521)
08-29 14:16:44.523 24258 24258 E EventBus:      at org.greenrobot.eventbus.EventBus.postToSubscription(EventBus.java:484)
08-29 14:16:44.523 24258 24258 E EventBus:      at org.greenrobot.eventbus.EventBus.postSingleEventForEventType(EventBus.java:461)
08-29 14:16:44.523 24258 24258 E EventBus:      at org.greenrobot.eventbus.EventBus.postSingleEvent(EventBus.java:434)
08-29 14:16:44.523 24258 24258 E EventBus:      at org.greenrobot.eventbus.EventBus.post(EventBus.java:306)
08-29 14:16:44.523 24258 24258 E EventBus:      at org.greenrobot.eventbus.EventBus.postSticky(EventBus.java:359)
08-29 14:16:44.523 24258 24258 E EventBus:      at com.telenav.osv.event.EventBus.postSticky(EventBus.java:44)
08-29 14:16:44.523 24258 24258 E EventBus:      at com.telenav.osv.service.UploadHandlerService$6.run(UploadHandlerService.java:291)
08-29 14:16:44.523 24258 24258 E EventBus:      at android.os.Handler.handleCallback(Handler.java:746)
08-29 14:16:44.523 24258 24258 E EventBus:      at android.os.Handler.dispatchMessage(Handler.java:95)
08-29 14:16:44.523 24258 24258 E EventBus:      at android.os.Looper.loop(Looper.java:148)
08-29 14:16:44.523 24258 24258 E EventBus:      at android.app.ActivityThread.main(ActivityThread.java:5443)
08-29 14:16:44.523 24258 24258 E EventBus:      at java.lang.reflect.Method.invoke(Native Method)
08-29 14:16:44.523 24258 24258 E EventBus:      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:728)
08-29 14:16:44.523 24258 24258 E EventBus:      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:618)
```

```
08-29 14:16:50.426 24258 25036 E Volley  : [30160] HurlStack.addBodyIfExists: Error writing request body to stream. thread interrupted
```

```
08-29 14:16:51.556 24258 25025 E Volley  : [30156] BasicNetwork.performRequest: Unexpected response code 400 for http://openstreetcam.org/1.0/photo/
```

```
08-29 14:16:51.789 24258 24343 D UploadManager: onRequestFinished: for video or photo file
08-29 14:16:51.808 24258 24343 D OsvRequestResponseListener: Volley error on request. Error: com.android.volley.ServerError
08-29 14:16:51.808 24258 24343 D OsvRequestResponseListener:    at com.android.volley.toolbox.BasicNetwork.performRequest(BasicNetwork.java:191)
08-29 14:16:51.808 24258 24343 D OsvRequestResponseListener:    at com.android.volley.NetworkDispatcher.run(NetworkDispatcher.java:133)
```

Of course, if OSC wasn't able to upload all the files, it should also not say that the upload was finished.

naoliv, 2018-08-29:
I also accept suggestion on what to do with this:

[image]

No matter what I try, it says that the upload was finished but doesn't remove the 2 tracks.
And it seems that if everything is not uploaded, it won't close and process the track in OSC site.

alexandrui-telenav, 2018-08-30:
@cameliao-telenav / @horatiuf , please advise

closed 2018-08-30 by alexandrui-telenav

mds08011, 2018-09-18:
I had a similar issue with empty tracks trying to upload.  I ended up going into my File Manager an deleting the odd 0MB photo files and associated track data.  This cleared my upload queue.