kartaview/android #144: Feature request: Orientations other than straight forward

myself248, 2018-09-21:
I was using OSC images to confirm some POI data recently, and found myself using Bing's 360 images because they showed storefronts better. The OSC images were all straight down the road and showed storefronts only obliquely.

I'd like to run OSC pointed out my side window, or possibly several instances (I have a bunch of phones/tablets that could run it), pointing left, right, and forward.

To do this and make the metadata correct, I'd like an "azimuth offset" setting on each device. Presets for 90/270 would be good for my proposed trio, but I could also imagine a 45/315 pair being really useful too.

alexandrui-telenav, 2018-10-01:
Hi @myself248 , the metadata stores the heading info of the phone, if all goes well the photos will be pointed correctly without a manual offset.
Have you tried doing a recording in the meantime?

mds08011, 2019-03-16:
Similar request: #84 

That user (james2432) actually implemented a "set bearing"/azimuth offset feature in his GUI Uploader: https://github.com/osmottawa/OSVUploadr if you want to give that a try.