kartaview/android #145: Uploading track now failing after closing app during upload

zayuim, 2018-12-28:
Hi I went out to record a track today. And while uploading it I closed the app and the upload stopped.
I went to reupload it, and now it is jumping to 100% Upload Completed. But the track is still in my local files (seemingly intact).

What's more, is that the app is now crashing when I view the track, although all the files (900MB+) are still in the apps data folder.

What can I do to upload the track?

brivars, 2019-01-13:
Looks like had exactly the same issue just today too!
It seems to have resolved itself though. After having the same symptoms described above, I tried pushing upload all several more times and it was always jumping to the 100% but I noticed that the total size there was getting smaller every time e.g. if it started out jumping to 100% 756MB/756MB the first time, then some time later it went down to say 245MB/245MB and lastly it was 100% 0/0MB and shortly after the track disappeared from the list in the app and showed up on the website as "Processing" so I'm assuming it was all successfully uploaded at some point.

Still a buggy behavior on app side though. Hope this description solves to get to the issue.

zayuim, 2019-01-13:
@brivars Yeah my problem fixed as well. All I had to do was record another track and upload that.