kartaview/android #146: Add Auto-center location feature

sanecito, 2019-01-12:
Problem:

Currently you have to manually adjust the map's focus (e.g. the map draws, and if you move say, 5 miles north, the map doesn't redraw the map so you remain at the center of the map). This makes it significantly harder to use the OSC app to see what streets haven't been captured yet when using the WayLens Horizon.

Basically for my case, I have a WayLens Horizon and want to be able to drive around using the OSC app and glance at OSC to see what streets I should drive to nearby to capture them. This becomes an extremely dangerous task if I have to pull over often to readjust the map manually.

Expected behaviors (based on patterns from other apps):

It should auto-center by default with the target icon being colored. If a user manually moves the map, this turns the auto-center toggle to off and greys out the target icon. Pressing the target icon toggles auto-center back on and turns the target icon back to colored.

mds08011, 2019-01-17:
I’d like to see this as well.  Essentially, the map would stay “locked” on your GPS location and follow you as you move?

sanecito, 2019-01-26:
@mds08011 Yep, exactly...similar to a navigation app.