kartaview/android #148: Auto-upload notification

brivars, 2019-01-15:
I tried switching on auto-upload at some point after I had some problems with manual one.
And as soon as the auto-upload started, the phone started dinging like crazy - there is a notification by default for every image that is being uploaded - so uploading one track is dinging constantly a thousand times. 
Of course, you can turn off notification sound for the app and then it is fine but this should not be the default app behavior, should it?

I'm on Mi A1 on Android One 8.10

mds08011, 2019-03-05:
Were you uploading over a mobile data connection or Wifi?  I have experienced this issue in the past but it seemed to only occur when uploading via mobile data.

horatiuf, 2019-03-06:
Hello @brivars,
Can you provide the app version on which you encountered this issue?

brivars, 2019-03-10:
Only ever uploaded via WiFi.
Not sure about the app version as it has updated since then. But it was the most recent version as of my OP - 15th January