kartaview/android #149: Can't find some dependencies

Rudloff, 2019-02-06:
Hello,

When trying to build 2.4.0 with `gradle assemble`, it can't find some dependencies:
```
Could not resolve all files for configuration ':app:mockDebugCompileClasspath'.
> Could not find com.facebook.network.connectionclass:connectivity-osc:1.0.0.
  Searched in the following locations:
      file:/home/pierre/www/android-sdk/extras/m2repository/com/facebook/network/connectionclass/connectivity-osc/1.0.0/connectivity-osc-1.0.0.pom
      file:/home/pierre/www/android-sdk/extras/m2repository/com/facebook/network/connectionclass/connectivity-osc/1.0.0/connectivity-osc-1.0.0.jar
      file:/home/pierre/www/android-sdk/extras/google/m2repository/com/facebook/network/connectionclass/connectivity-osc/1.0.0/connectivity-osc-1.0.0.pom
      file:/home/pierre/www/android-sdk/extras/google/m2repository/com/facebook/network/connectionclass/connectivity-osc/1.0.0/connectivity-osc-1.0.0.jar
      file:/home/pierre/www/android-sdk/extras/android/m2repository/com/facebook/network/connectionclass/connectivity-osc/1.0.0/connectivity-osc-1.0.0.pom
      file:/home/pierre/www/android-sdk/extras/android/m2repository/com/facebook/network/connectionclass/connectivity-osc/1.0.0/connectivity-osc-1.0.0.jar
      https://jcenter.bintray.com/com/facebook/network/connectionclass/connectivity-osc/1.0.0/connectivity-osc-1.0.0.pom
      https://jcenter.bintray.com/com/facebook/network/connectionclass/connectivity-osc/1.0.0/connectivity-osc-1.0.0.jar
      https://dl.google.com/dl/android/maven2/com/facebook/network/connectionclass/connectivity-osc/1.0.0/connectivity-osc-1.0.0.pom
      https://dl.google.com/dl/android/maven2/com/facebook/network/connectionclass/connectivity-osc/1.0.0/connectivity-osc-1.0.0.jar
  Required by:
      project :app
> Could not find com.telenav.ffmpeg:ffmpeg:1.0.4.
  Searched in the following locations:
      file:/home/pierre/www/android-sdk/extras/m2repository/com/telenav/ffmpeg/ffmpeg/1.0.4/ffmpeg-1.0.4.pom
      file:/home/pierre/www/android-sdk/extras/m2repository/com/telenav/ffmpeg/ffmpeg/1.0.4/ffmpeg-1.0.4.jar
      file:/home/pierre/www/android-sdk/extras/google/m2repository/com/telenav/ffmpeg/ffmpeg/1.0.4/ffmpeg-1.0.4.pom
      file:/home/pierre/www/android-sdk/extras/google/m2repository/com/telenav/ffmpeg/ffmpeg/1.0.4/ffmpeg-1.0.4.jar
      file:/home/pierre/www/android-sdk/extras/android/m2repository/com/telenav/ffmpeg/ffmpeg/1.0.4/ffmpeg-1.0.4.pom
      file:/home/pierre/www/android-sdk/extras/android/m2repository/com/telenav/ffmpeg/ffmpeg/1.0.4/ffmpeg-1.0.4.jar
      https://jcenter.bintray.com/com/telenav/ffmpeg/ffmpeg/1.0.4/ffmpeg-1.0.4.pom
      https://jcenter.bintray.com/com/telenav/ffmpeg/ffmpeg/1.0.4/ffmpeg-1.0.4.jar
      https://dl.google.com/dl/android/maven2/com/telenav/ffmpeg/ffmpeg/1.0.4/ffmpeg-1.0.4.pom
      https://dl.google.com/dl/android/maven2/com/telenav/ffmpeg/ffmpeg/1.0.4/ffmpeg-1.0.4.jar
  Required by:
      project :app
> Could not find uk.co.senab.photoview:photoview-osc:1.0.0.
  Searched in the following locations:
      file:/home/pierre/www/android-sdk/extras/m2repository/uk/co/senab/photoview/photoview-osc/1.0.0/photoview-osc-1.0.0.pom
      file:/home/pierre/www/android-sdk/extras/m2repository/uk/co/senab/photoview/photoview-osc/1.0.0/photoview-osc-1.0.0.jar
      file:/home/pierre/www/android-sdk/extras/google/m2repository/uk/co/senab/photoview/photoview-osc/1.0.0/photoview-osc-1.0.0.pom
      file:/home/pierre/www/android-sdk/extras/google/m2repository/uk/co/senab/photoview/photoview-osc/1.0.0/photoview-osc-1.0.0.jar
      file:/home/pierre/www/android-sdk/extras/android/m2repository/uk/co/senab/photoview/photoview-osc/1.0.0/photoview-osc-1.0.0.pom
      file:/home/pierre/www/android-sdk/extras/android/m2repository/uk/co/senab/photoview/photoview-osc/1.0.0/photoview-osc-1.0.0.jar
      https://jcenter.bintray.com/uk/co/senab/photoview/photoview-osc/1.0.0/photoview-osc-1.0.0.pom
      https://jcenter.bintray.com/uk/co/senab/photoview/photoview-osc/1.0.0/photoview-osc-1.0.0.jar
      https://dl.google.com/dl/android/maven2/uk/co/senab/photoview/photoview-osc/1.0.0/photoview-osc-1.0.0.pom
      https://dl.google.com/dl/android/maven2/uk/co/senab/photoview/photoview-osc/1.0.0/photoview-osc-1.0.0.jar
  Required by:
      project :app
> Could not find com.telenav.streetview.scalablevideoview:scalablevideoview-osc:1.0.0.
  Searched in the following locations:
      file:/home/pierre/www/android-sdk/extras/m2repository/com/telenav/streetview/scalablevideoview/scalablevideoview-osc/1.0.0/scalablevideoview-osc-1.0.0.pom
      file:/home/pierre/www/android-sdk/extras/m2repository/com/telenav/streetview/scalablevideoview/scalablevideoview-osc/1.0.0/scalablevideoview-osc-1.0.0.jar
      file:/home/pierre/www/android-sdk/extras/google/m2repository/com/telenav/streetview/scalablevideoview/scalablevideoview-osc/1.0.0/scalablevideoview-osc-1.0.0.pom
      file:/home/pierre/www/android-sdk/extras/google/m2repository/com/telenav/streetview/scalablevideoview/scalablevideoview-osc/1.0.0/scalablevideoview-osc-1.0.0.jar
      file:/home/pierre/www/android-sdk/extras/android/m2repository/com/telenav/streetview/scalablevideoview/scalablevideoview-osc/1.0.0/scalablevideoview-osc-1.0.0.pom
      file:/home/pierre/www/android-sdk/extras/android/m2repository/com/telenav/streetview/scalablevideoview/scalablevideoview-osc/1.0.0/scalablevideoview-osc-1.0.0.jar
      https://jcenter.bintray.com/com/telenav/streetview/scalablevideoview/scalablevideoview-osc/1.0.0/scalablevideoview-osc-1.0.0.pom
      https://jcenter.bintray.com/com/telenav/streetview/scalablevideoview/scalablevideoview-osc/1.0.0/scalablevideoview-osc-1.0.0.jar
      https://dl.google.com/dl/android/maven2/com/telenav/streetview/scalablevideoview/scalablevideoview-osc/1.0.0/scalablevideoview-osc-1.0.0.pom
      https://dl.google.com/dl/android/maven2/com/telenav/streetview/scalablevideoview/scalablevideoview-osc/1.0.0/scalablevideoview-osc-1.0.0.jar
  Required by:
      project :app
> Could not find com.android.volley:volley-osc:1.0.1.
  Searched in the following locations:
      file:/home/pierre/www/android-sdk/extras/m2repository/com/android/volley/volley-osc/1.0.1/volley-osc-1.0.1.pom
      file:/home/pierre/www/android-sdk/extras/m2repository/com/android/volley/volley-osc/1.0.1/volley-osc-1.0.1.jar
      file:/home/pierre/www/android-sdk/extras/google/m2repository/com/android/volley/volley-osc/1.0.1/volley-osc-1.0.1.pom
      file:/home/pierre/www/android-sdk/extras/google/m2repository/com/android/volley/volley-osc/1.0.1/volley-osc-1.0.1.jar
      file:/home/pierre/www/android-sdk/extras/android/m2repository/com/android/volley/volley-osc/1.0.1/volley-osc-1.0.1.pom
      file:/home/pierre/www/android-sdk/extras/android/m2repository/com/android/volley/volley-osc/1.0.1/volley-osc-1.0.1.jar
      https://jcenter.bintray.com/com/android/volley/volley-osc/1.0.1/volley-osc-1.0.1.pom
[... 26 lines cut ...]
      file:/home/pierre/www/android-sdk/extras/android/m2repository/com/matthewtamlin/dotindicator/dotindicator-osc/1.0.0/dotindicator-osc-1.0.0.jar
      https://jcenter.bintray.com/com/matthewtamlin/dotindicator/dotindicator-osc/1.0.0/dotindicator-osc-1.0.0.pom
      https://jcenter.bintray.com/com/matthewtamlin/dotindicator/dotindicator-osc/1.0.0/dotindicator-osc-1.0.0.jar
      https://dl.google.com/dl/android/maven2/com/matthewtamlin/dotindicator/dotindicator-osc/1.0.0/dotindicator-osc-1.0.0.pom
      https://dl.google.com/dl/android/maven2/com/matthewtamlin/dotindicator/dotindicator-osc/1.0.0/dotindicator-osc-1.0.0.jar
  Required by:
      project :app
> Could not find com.crashlytics.sdk.android:crashlytics-ndk:1.1.6.
  Searched in the following locations:
      file:/home/pierre/www/android-sdk/extras/m2repository/com/crashlytics/sdk/android/crashlytics-ndk/1.1.6/crashlytics-ndk-1.1.6.pom
      file:/home/pierre/www/android-sdk/extras/m2repository/com/crashlytics/sdk/android/crashlytics-ndk/1.1.6/crashlytics-ndk-1.1.6.jar
      file:/home/pierre/www/android-sdk/extras/google/m2repository/com/crashlytics/sdk/android/crashlytics-ndk/1.1.6/crashlytics-ndk-1.1.6.pom
      file:/home/pierre/www/android-sdk/extras/google/m2repository/com/crashlytics/sdk/android/crashlytics-ndk/1.1.6/crashlytics-ndk-1.1.6.jar
      file:/home/pierre/www/android-sdk/extras/android/m2repository/com/crashlytics/sdk/android/crashlytics-ndk/1.1.6/crashlytics-ndk-1.1.6.pom
[... cut ...]
      https://jcenter.bintray.com/com/crashlytics/sdk/android/crashlytics-ndk/1.1.6/crashlytics-ndk-1.1.6.jar
      https://dl.google.com/dl/android/maven2/com/crashlytics/sdk/android/crashlytics-ndk/1.1.6/crashlytics-ndk-1.1.6.pom
      https://dl.google.com/dl/android/maven2/com/crashlytics/sdk/android/crashlytics-ndk/1.1.6/crashlytics-ndk-1.1.6.jar
  Required by:
      project :app
```

I see some of these are in local folders but it can't find them for some reason.
Am I missing some prebuild step?

Zahnstocher, 2019-03-10:
I was able to build it.
But it is not possible to build OpenStreetCam without heavy modifications and removing many dependencies.

ithink314, 2019-04-19:
Where did you find source for version 2.4.0? Latest available seems to be 2.1.6, while 2.7.3 is on play store?

mds08011, 2019-08-06:
> Where did you find source for version 2.4.0?

2.4.0 is in Github releases: https://github.com/openstreetcam/android/releases/tag/2.4.0

Zahnstocher, 2019-08-09:
> 2.4.0 is in Github releases: https://github.com/openstreetcam/android/releases/tag/2.4.0

2.4.0 is without source code, this is also mentioned in the description:
> // Source code is not yet up to date.

Last version with source code is 2.1.6.

If you look here, you can also see that the commit id does not change after 2.1.6, it is always 4f20e5e:
https://github.com/openstreetcam/android/releases

Also last commit is from Dec 22, 2017
https://github.com/openstreetcam/android/commits/master
but version 2.4.0 is from 23 Aug 2018.

So there is no source code available after version 2.1.6.

mds08011, 2019-08-09:
>2.4.0 is without source code, this is also mentioned in the description

I think the description is referring to the code in the main Github repository.  However, the 2.4.0 release is packaged with 3 files: “osc_2.4.0_release.apk”, “Source code (zip)”, and “Source code (tar.gz)”.  

Have you looked at the source code packaged with the release, or only the date of Github commits?

Zahnstocher, 2019-08-10:
> Have you looked at the source code packaged with the release, or only the date of Github commits?

Yes, they refer to the commit 4f20e5e.
I think they are auto generated from this commit.
The content of the 3 source codes packages (zip and tar.gz) are binary identical (2.1.6, 2.3, 2.4.0).
They are all 2.1.6.

manhlhvm, 2020-11-26:
i same error with version 4.0. solution for this ?