kartaview/android #151: Lost local files when putting osc on background

alex9099, 2019-06-07:
Hi,
I was uploading a trip i did by bike, but when i was uploading i opened another app and when i  tried to return to osc it crashed and now says that i don't have any local files? Does this mean that i lost all of my trip?

alex9099, 2019-06-07:
[log_06_07_11_54.txt](https://github.com/openstreetcam/android/files/3265952/log_06_07_11_54.txt)
Here is the log file

alex9099, 2019-06-07:
It seems like it took a long while to be recognized on the website, i am not sure if it sent the whole thing but anyway :D

closed 2019-06-07 by alex9099

mirunac-telenav, 2019-06-11:
@alex9099 - an unfortunate storage downtime occurred and it took a while to get it resolved. Everything is up and running now.

alex9099, 2019-06-11:
Was that downtime 4 days ago? The upload went through, it just took a long while to appear on the site

mirunac-telenav, 2019-06-12:
@alex9099 - yep, started over the weekend.