kartaview/android #153: "Source code is not yet up to date"?

vpzomtrrfrt, 2019-08-03:
Why is there no source code published here since December 2017?

ithink314, 2019-08-06:
(Following) It seems this code is now closed proprietary source, although they still advertise: "free and open platform."

tuxayo, 2019-08-19:
For the record, the latest libre/open source release is the v2.1.6
https://github.com/openstreetcam/android/tree/2.1.6

tuxayo, 2019-08-19:
@Telenav @openstreetcam This is important, the only reason I was using and advertising OpenStreetCam over Mapillary was that the apps and the website where libre.

It's weird to require users to use closed tools to work, create and collect libre/open data.

PanderMusubi, 2019-08-19:
So we close this issue on closedstreetcam?

davidak, 2019-08-19:
I asked them to clearify the situation via E-Mail and Twitter.

alexandrui-telenav, 2019-08-20:
The Android repo is indeed way behind development, same for [openstreetcam.org](https://github.com/openstreetcam/openstreetcam.org). Others like [upload-scripts](https://github.com/openstreetcam/upload-scripts) and [josm-plugin](https://github.com/openstreetcam/josm-plugin) are doing better but still not great. We had to focus lately on some fundamental tech aspects on the platform and maintaining the open source part got deprioritized in the end.

In the coming weeks we’ll have code updates on this repo and others, we don’t have any intention of keeping the platform closed. Sorry for not being as active here as we should've.

alexandrui-telenav, 2019-08-20:
Let's keep this issue open until actual code update.

> So we close this issue on closedstreetcam?

Let's keep it open until actual code update.

PanderMusubi, 2019-08-20:
Thanks for the update with good news.

davidak, 2019-08-20:
>maintaining the open source part got deprioritized in the end.

What's the problem to `git push` at the end of the workday? Then we see at least that the project is not dead.

532910, 2019-08-20:
https://github.com/openstreetcam/android/issues/153#issuecomment-522945583 --- @telenav replies that it's still opensource.

alexandrui-telenav, 2019-08-21:
> > maintaining the open source part got deprioritized in the end.
> 
> What's the problem to `git push` at the end of the workday? Then we see at least that the project is not dead.

Clean up / small refactoring for external SDK usage, API keys etc. Nothing major but with summer holidays and a small team I can't promise something very very fast. But point well taken.

Project's clearly not dead as we update the app on Google Play quite regularly, it's the Github part we have to do a better job at..

ec1oud, 2019-08-31:
I don't use Google Play Store.  So that means the latest is still 2.4.0, a year old.  Did any improvements get made in the last year?

ithink314, 2019-09-03:
>Sorry for not being as active here as we should've.

Going on 2 years since last update, this sounds hollow. Sorry for not being active with my Waylens camera (proprietary) or android app.

mds08011, 2019-10-02:
> 
> 
> I don't use Google Play Store. So that means the latest is still 2.4.0, a year old. Did any improvements get made in the last year?

@ec1oud the 2.9.1 .apk is in releases now: https://github.com/openstreetcam/android/releases

camelCaseNick, 2019-11-15:
> Clean up / small refactoring for external SDK usage, API keys etc. Nothing major but with summer holidays and a small team I can't promise something very very fast. But point well taken.
>
> Project's [...]

You published the 2.9.1 apk with old source code and nothing is happening in the repo? Why do you care so little about open source, that not even a yearly commit is possible?

siliconninja, 2019-11-21:
Check the `release_2.9.1_github` branch. I see some updated source code there, the new branch just wasn't announced in this issue. It's not in the default `master` branch. Thank you for updating the source code!

zekooooo, 2020-01-06:
Hello, have there been updates? I would really like to have OpenStreetCam on F-Droid.

ithink314, 2020-01-08:
@zekooooo 3.0.1 is on play store; only 2.9.1 is on github. It probably won't ever make it to f-droid.

tuxayo, 2020-02-20:
v3.0.1 source code seems to have been released not in the repo but in the [release page](https://github.com/openstreetcam/android/releases) as a source code archive.

Zahnstocher, 2020-02-20:
> v3.0.1 source code seems to have been released not in the repo but in the [release page](https://github.com/openstreetcam/android/releases) as a source code archive.

No, this is the very old 2.1.6 code (end of 2017).
It is not even the latest available 2.9.1 code:
https://github.com/openstreetcam/android/tree/release_2.9.1_github

tuxayo, 2020-02-25:
`build.grade` =>  `appVersionName            : "2.1.6",`

Ho damn, you are right :(

Yesterday, I just recommended OpenStreetCam at a mapping party just because I though it was libre again T_T

tuxayo, 2020-06-11:
Version 3.1.6 just released and it's not libre/open source.

Latest one (2.1.6) has version code 86. So by using [Aurora Store](https://f-droid.org/en/packages/com.aurora.store/) or [Yalp Store](https://f-droid.org/en/packages/com.github.kiliakin.yalpstore/) it's possible to get it. (menu => manual download => input the version code)

Hopefully version 2.1.6 will still work.

Bibi56, 2020-06-19:
On http://blog.improveosm.org/en/2019/12/openstreetcam-and-improveosm-are-moving-to-grab/ (Dec 2019 about ImproveOSM and OpenStreetCam):

> Open Source: Good news, the code license is changing from LGPL to the more permissive MIT license. The imagery license will remain the same, CC BY-SA version 4. 

So Open Source, but with a minor defect: the code is missing.

EchedeyLR, 2020-07-30:
Then is not Open Source. The Open Source definition that the software, itself, must be Free Software by design --> https://opensource.org/osd

EchedeyLR, 2020-07-30:
> Version 3.1.6 just released and it's not libre/open source.
> 
> Latest libre/open source one (2.1.6) has version code 86. So by using [Aurora Store](https://f-droid.org/en/packages/com.aurora.store/) or [Yalp Store](https://f-droid.org/en/packages/com.github.kiliakin.yalpstore/) it's possible to get it. (menu => manual download => input the version code)
> 
> Hopefully version 2.1.6 will still work.

You are not the only one...

Bibi56, 2020-09-03:
@Extarys, sad yes but the license until 12 Dec 2019 was _GNU LESSER GENERAL PUBLIC LICENSE_ then changed to _MIT License_.

_GNU LESSER GENERAL PUBLIC LICENSE_, valid until 3.0.0 at least states:
_The "Minimal Corresponding Source" for a Combined Work means the
Corresponding Source for the Combined Work, excluding any source code
for portions of the Combined Work that, considered in isolation, are
based on the Application, and not on the Linked Version._

So it is not supposed to be closed code at least until 3.0.0.

For 3.0.1 and following versions, as it is using a _GNU LESSER GENERAL PUBLIC LICENSE_ for its base, IMHO - but I'm not a jurist - the derivative work like 3.1.6 should comply to the previous statement.
So **at least** we are missing the 3.0.0 source code here.

Extarys, 2020-09-04:
@Bibi56 What I meant was the current source code in the repo is old (v2.9.1) compared to the current APK. So the latest version is not open source no matter what the license says right now because there is no source.
Also, I discovered this yesterday so I didn't have time to investigate the "new" company behind this and make my mind about their reputation and if they should be trusted.

They bought this at the end of last year, so maybe* they'll catch up soon and continue their Open-Source works :crossed_fingers:

chip-grab, 2020-09-04:
@Extarys, I have an insight on this that @grab will continue open source work. 😉
Updates to come soon. Stay tunned. 🤞

davidak, 2020-09-04:
@chip-grab do you have an estimate for a release? we are waiting for 3 years...

zekooooo, 2020-09-06:
Happy to hear this @CHip especially in the light of Facebook buying Mapillary.

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

mnalis, 2021-01-19:
@chip-grab has there been any progress on making the app open-source? If I read it correctly, [build.gradle](https://github.com/kartaview/android/blob/master/build.gradle#L9) still requires proprietary components in `4.0` ?

tuxayo, 2021-02-17:
The source code of the repo and the archive from the [releases](https://github.com/kartaview/android/releases/tag/4.0) is still the one of version 2.1.6 (version code 86).

So version 4.0 is still not libre/open source

Zahnstocher, 2021-02-17:
> The source code of the repo and the archive from the [releases](https://github.com/kartaview/android/releases/tag/4.0) is still the one of version 2.1.6 (version code 86).
> 
> So version 4.0 is still not libre/open source

It is version 4.0, version code 118:
https://github.com/kartaview/android/blob/master/build.gradle#L134
It was released on 2 Dec 2020:
https://github.com/kartaview/android/commit/ae5b56b80de596612c4f0bea0b54c6e615d21966
I tried to build it a few days later, but I failed.
Some things in the code are not defined and it heavily depends on third party libs, some are closed source, some are not available in the repositories and I had to find them myself.
Then I found out, that I need, API key for Mapbox, so I registered there.
For Google (analytics or something like that), I created a dummy API key file.
Then I found out, that I need Skobbler API keys, which I don't have. Why is Skobbler dependency still there, I thought it was replaced by the new Mapbox dependency...
After several hours I gave up, it was just a waste of time. :cry:
I think this open source project is completely broken.
I wonder if the developer who committed this code ever tried to build it. :confused:

hamishmb, 2022-10-11:
Bumping again. I'm happy to be of assistance, when I have time and spare executive function (it varies when you have autism).

I could help get this release building, too, and try to get in contact with the devs again, but I can't do these things by myself in any reasonable timescale. Shall we start coordinating something?

chrisdebian, 2022-10-18:
> Bumping again. I'm happy to be of assistance, when I have time and spare executive function (it varies when you have autism).
>
> I could help get this release building, too, and try to get in contact [...]

Hamish, I've just opened a new issue, asking for the code and source to be uploaded, as this is a requirement of the OS licence, and will enable people to fork, should they want to review or modify the code.

chrisdebian, 2023-02-12:
I received an update, the other day; ver 5.5.1 (150), but I can't see this release on GitHub. Could someone else check, in case I'm being paranoid about open source licenses being contravened.

Thanks,

Chris
chris_debian

EchedeyLR, 2023-02-12:
Hi, this app is not FLOSS anymore sadly.

hamishmb, 2023-04-04:
Unfortunately you're probably right. Unless the server side code is open, this might be a bit of a lost cause.

mnalis, 2023-08-21:
> I received an update, the other day; ver 5.5.1 (150), but I can't see this release on GitHub. Could someone else check, in case I'm being paranoid about open source licenses being contravened.

Hmm, the code was GPL3 until 2019, when https://github.com/kartaview/android/commit/502b9004eb7ab4f9e67f49f965e560828b0f1210 changed it to MIT license (which is not copyleft, thus cannot guarantee that the code will stay open) ?!

Have all people who ever contributed to the code agreed to this license change? If not, this license change is illegal. 

Call to people: if you ever contributed to code of what is now Kartaview, have not given up / transfered you copyright, and disagree with this license change, please comment here or contact me.

davidak, 2023-08-23:
@mnalis you can see contributors at https://github.com/kartaview/android/graphs/contributors.

People that contributed before the license change are @james2432, @vdechenaux and @Xevib.

mnalis, 2023-08-24:
> @mnalis you can see contributors at https://github.com/kartaview/android/graphs/contributors.

@davidak thanks, that indeed lists some of them (and if any of them owns copyright on part of the code and didn't authorize license change I hope they will reply here), however:

- there are even more contributors produced by `git log --before '2019-12-12' | grep '^Author:' | sort -u`
- I'm not sure if git commit list all contributors (depending on contribution, i.e. if someone sent patch by e-mail etc.)