kartaview/android #155: Recording had to be stopped

panthersdude2001, 2019-10-05:
When recording track, it says that Recording had to be stopped.
I tried clearing cache and data, but still does not work.

panthersdude2001, 2019-10-05:
[image]
This is the error I am getting

panthersdude2001, 2019-10-05:
Android Phone is 6.0.1
OSC is 2.9.1 (104)
Model of Phone is SM-G550T

panthersdude2001, 2019-10-05:
Nevermind, got it working after repairing SD card.

closed 2019-10-05 by panthersdude2001

brmonaghan, 2021-06-18:
Please reopen this as my account was deleted.
[image]
Still not working as its still doing it even without SD Card present. Link2SD moved KV to SD Card and the application will NOT save tracks to SD Card by default.