kartaview/android #157: OpenStreetCam stuck on processing?

panthersdude2001, 2020-02-24:
Every sequence I took this weekend was successfully uploaded this morning and are now processing.
But they seem to be that way for about 2 hours.
Is there a bug with this, or is there a higher than normal processing time for you guys?
Thanks

tuxayo, 2020-02-25:
24h ago with someone at a mapping party, we used OSC and the tracks are still processing.

@panthersdude2001 What were the typical process times that you had in the past?

panthersdude2001, 2020-02-25:
https://i.imgur.com/Q5ojmDz.png
Still processing...
It takes about 30 minutes to process sequences.

panthersdude2001, 2020-02-25:
The leaderboard seems to be not working as well.
[image]

panthersdude2001, 2020-03-01:
After 7 days, the sequences are still processing...

panthersdude2001, 2020-03-02:
@chip-grab any word on when processing will be done? this seems to be a bug with the processing servers.

panthersdude2001, 2020-03-03:
Looks like the processing bug has gone away for some, not all. Closing at it's no longer processing.

chip-grab, 2020-03-03:
@panthersdude2001, @tuxayo, indeed we have some issues on processing pipeline. We are working on a fix, and come back to you. Thank you for your patience.

panthersdude2001, 2020-03-03:
Thanks for identifying the issue @chip-grab ! I will check back later to see if my tracks are processing. Another user is starting to get their tracks processed as well (went down from 75 to 73 today) Sent from Mail for Windows 10 From: CHipSent: Tuesday, March 3, 2020 9:51 AMTo: openstreetcam/androidCc: Bryce Monaghan; MentionSubject: Re: [openstreetcam/android] OpenStreetCam stuck on processing? (#157) @panthersdude2001, @tuxayo, indeed we have some issues on processing pipeline. We are working on a fix, and come back to you. Thank you for your patience.

panthersdude2001, 2020-03-03:
About time! Now down to 20 processing.
But, there is no points assigned...  @chip-grab is this going to get fixed soon?
[image]

panthersdude2001, 2020-03-04:
Down to 14, now with points! Closing down issue as this issued has now been resolved.
[image]

closed 2020-03-04 by panthersdude2001