kartaview/android #181: Has development stopped?

LoveIsGrief, 2022-05-30:
The repo was last updated a long time ago yet the android app was updated just this month.

chrisdebian, 2022-05-30:
I've made this point before, but updating the repo seems very low priority.

Chris
chris_debian

LoveIsGrief, 2022-06-03:
I'm not sure what that means. I just discovered this and have no history of what went down. A link in the README could help, or if this opensource version has been abandoned, archiving the repo would help indicate that.

joeybab3, 2022-08-03:
It means they are not focused on keeping the repo up to date with what's actually published. They are still working on the project but they work primarily behind the scenes. Even getting support for platform-breaking issues can sometimes take months, it's not abandoned but the repo not likely to be updated any time soon.

chrisdebian, 2022-08-03:
I'm unclear about the way forward. As a long time (1998) user/ sys admin/ Information Architect, I know the importance of open source software, and the role of the developer AND the user.

If the developers aren't sufficiently resourced to publish the source code, then it's worrying because they could potentially be contravening the open source license. 

I feel like I am contributing a lot of imagery, potentially for financial gain, by Grab, and also use by OSM, and it feels like the contributors are being forgotten about. I suspect like OpenStreetCam and OpenStreetView, and now Kartaview, this software is likely to be forked,

https://wiki.openstreetmap.org/wiki/KartaView

and users will move (using automated tooling, again) contributions to a new repository, like they did when Mapillary was acquired by Facebook.

Kartaview contributors are content for imagery to be used for commercial gain, but I think most are motivated by the support the imagery gives to OSM. It feels like the community is losing patience.

Can Grab/ Kartaview please clarify the current situation, so the contributors can make a decision about whether they should continue to support this endeavour?

I'm disappointed to be writing this, but some clarification is needed.

Chris
chris_debian

joeybab3, 2022-08-03:
I've gone through this process with them like 3 or 4 times... Eventually, someone sees and responds and we hear about how the major rework is just around the corner, this time for real, and then they go dark for 6 months until the processing server crashes again and they are forced to come back for 15 minutes. 

The app doesn't work on my pixel 6 and they killed the waylens project during one fo the acquisitions so as it stands I have no longer been able to contribute.

It sucks considering I've uploaded something like 43000mi/70000km worth of imagery but at this point, I can't even tell you what that number is because the site doesn't actually work anymore!
[image]

Our options are to start tagging random people who have in the past responded, and maybe get a response and have a little back and forth dialogue for a day or so then return to nothing or give up.

chrisdebian, 2022-08-03:
I just checked, and I can see my contributions:

https://kartaview.org/user/chris_debian

but besides that, I agree with your comments.

Chris

joeybab3, 2022-08-03:
Yeah I forgot that it takes a while to show anything on my profile

LoveIsGrief, 2022-08-03:
Yeah, these responses really don't make me want to create an account and contribute any data. It seems like "opensource" is kind of loosely applies here without much transparency.

retiolus, 2022-09-12:
Since there is no answer, perhaps what we should do is find out how to export our data to preserve it for the next project or to put it online somewhere else.

chrisdebian, 2022-09-12:
@retiolus I get what you're saying, but am not sure whether a similar project exists, that could better maintain and publish the open source software, and have a use for the imagery. Removing the commercial element, is it possible, I wonder, to just use the imagery for OSM and associated projects. If the commercial element isn't working, can we just forget about it?

Anyone have any thoughts?

Chris
chris_debian

joeybab3, 2022-09-12:
There was https://github.com/Streets-Data-Collaborative/osc-tools but unfortunately it appears I was the last one to commit to it and it hasn't been updated for the new site... I can look into it for archival purposes this week but I would really appreciate a staff member/maintainer responding to us...

joeybab3, 2022-09-12:
@miruna-craciun @zhangtiedong0912 @chip-grab @zhaitianduo If any of you guys are still active on this project and able to respond to user concerns we would appreciate it.

chrisdebian, 2022-09-12:
@joeybab3 Good call. Maybe after a period, the main OSM mailing lists could be contacted, and this issue described briefly. This could be a way to raise awareness and give various stakeholders a chance to respond. 

The submitted data is open, and commercial entities are free to use it, but the community also expects code to be published and to maintain a relationship with the community.

Grab, please reach out, you can make this better.

Chris
chris_debian

hamishmb, 2022-09-14:
Personally I've just downgraded to the last open source release, as I'm slowly purging non free code from my phone, as far as possible at least.

Contacting the OSM mailing list seems like a good last resort. It's been over a year now since anything was published here, but the app has seen updates, so it seems worth doing.

chrisdebian, 2022-09-14:
Is anyone in this conversation from Grab/ developers?

Chris
chris_debian

chrisdebian, 2022-09-14:
> Personally I've just downgraded to the last open source release

Was that the last version they released, rather than the one on Play?

Cheers,

Chris
chris_debian

joeybab3, 2022-09-14:
> Is anyone in this conversation from Grab/ developers?
> 
> Chris chris_debian

The people I tagged above are those I know that have/currently work on the project. Last time I got someone to respond by borderline harassing grab on twitter but I'd prefer not to do that again.

chrisdebian, 2022-09-14:
Completely understand @joeybab3 ; I noted that you'd referenced earlier contacts. It just seems weird that the company/ developers aren't communicating.

joeybab3, 2022-09-14:
I've been contributing to this project for several years, for them to go months with major issues and not respond is quite par for course at this point. At least the server is still running, they had an issue where tracks didn't process for well over a month and there is no way to contact them when that happens.

hamishmb, 2022-09-14:
Yes, by "the last open source release" I meant the latest one available from the releases section of this repo.

LoveIsGrief, 2022-10-11:
I'm not sure what that means. I discovered this app yesterday while looking for an opensource alternative to Mapillary which was bought by Facebook. This app isn't on F-Droid, so a little harder to retrieve by degoogled phones.

If the app isn't opensource anymore, maybe the repo should be archived. A link to the stuff you "said before" might be useful in the read me regardless of archive status, since new people like me have no idea what's going on.

joeybab3, 2022-10-11:
> I'm not sure what that means. I discovered this app yesterday while looking for an opensource alternative to Mapillary which was bought by Facebook. This app isn't on F-Droid, so a little harder to [...]

There's not much to say/link. There are lots of short 1-5 comment dialogue bits spread around the various repos they have here and it all nets the same result. We've asked them(grab/telenav) to help us/update/fix things and they claim they will and then nothing happens.

If you don't want to use google play then go to the releases section of this repo and download the apk, but everyone responding to you here is just a user, we can't make them put it on FDroid or update this repo.

miruna-craciun, 2022-10-12:
To put it simply, no, development hasn’t stopped for KartaView. Historically speaking, we have addressed numerous requests here on Github (especially where we noticed an interest on the community's end), and have always been focused on improvements our community sought. Naturally, there are certain aspects that require a larger bandwidth, which we don’t have at the moment.
However, if we see a desire to replicate parts of KartaView in your projects or support compilation initiatives, please reach out to us and we’ll act on it. Additionally, if there are features or tools that might spike interest but you need backup on set-up or handling certain requests, do reach out to geo.kartaview@grabtaxi.com.

chrisdebian, 2022-10-12:
@miruna-craciun 

Thank you for taking the time to reach out and respond to this thread. Simply put, I think the community would be somewhat happier if you (grabtaxi) could regularly 'push'/ publish the code for the latest release of the Kartaview application. This is a basic requirement of most open source licenses. If this isn't done, you (grabtaxi) are potentially in breach of the license, and are alienating the community that provides the imagery data that you want. I think it's just a question of being open. 

These are just my thoughts, and I don't profess to represent the user community at large.

Thanks, again.
Chris
chris_debian

retiolus, 2022-10-14:
Anyone has a solution to export Kartaview data for the moment?

hamishmb, 2022-10-15:
I don't, but haven't tried.

For the moment, I have decided to uninstall it, until the situation improves - I never open it anyway.

LoveIsGrief, 2022-10-15:
It would be ironic if we had to reverse engineer the API of a supposedly opensource app, wouldn't it?

hamishmb, 2022-10-15:
At an undefined point in the future, when I have time and some more relevant-to-android skills, I'd be up for that.

Best solution is really to avoid the duplication of effort if possible, though.

chrisdebian, 2022-12-29:
@brmonaghan Bryce, could you DM me, I couldn't see how to.

Cheers,

Chris

chrisdebian, 2022-12-29:
Ahh, maybe it's not possible :-). I'll try to post my thoughts here.

Chris

chrisdebian, 2023-04-09:
Hi, all.

Just read about Mapilio, based on the open source code of Mapillary. Now to work out a way to move my imagery from KV.

mapilio
[https://mapilio.com/app](https://translate.google.com/website?sl=auto&tl=EN&hl&u=https://mapilio.com/app)
[https://mapilio.com/terms](https://translate.google.com/website?sl=auto&tl=EN&hl&u=https://mapilio.com/terms)
[https://mapilio.com/licenses](https://translate.google.com/website?sl=auto&tl=EN&hl&u=https://mapilio.com/licenses)
It is an app that is based on the open source of Mapillary. It is a project that is starting so it is promising, but it does not have many photos in its repository

Fingers crossed.

Chris

chrisdebian, 2023-04-09:
@brmonaghan Are you installing from Play?

https://play.google.com/store/apps/details?id=com.mapilio.app

LoveIsGrief, 2023-04-10:
Mapilio hasn't opensourced their server code from what I can see. https://github.com/mapilio  
The important thing they opensourced is their [mapilio-kit](https://github.com/mapilio/mapilio-kit).

> Mapilio Kit is a library for processing and uploading images to [Mapilio](https://www.mapilio.com/).

It doesn't seem like a better option than Mapillary or any other proprietary service.

chrisdebian, 2023-04-10:
It's maintained.

Chris

mnalis, 2023-04-12:
> It's maintained.

What is maintained? 

Unless I'm missing something, https://github.com/orgs/mapilio/repositories does not even have a repository for android code? Kartaview android code in this repository seems outdated and has other problems as noted, but the Mapilio does not seem to have **anything** published? Surely _"nothing"_ is even _**worse**_ that _"outdated"_ ?

chrisdebian, 2023-04-13:
Hi, all. 

I've had some clarification from the Mapilio Devs.

https://forum.mapilio.com/t/where-is-the-open-source-mapilio-android-app-hosted/112/2

In essence, because the app is still in beta, they are not yet able to publish the code for the Android app. My main concern is that the imagery is covered by the Creative Commons license, and ownership remains with the person who captured the imagery, so this is a good thing.

I'm encouraged by what Mapilio is doing, but I can see why contributors may stick with Mapillary and KV. I'll monitor the situation, and see how it is giving back to the community. For me, I need an app that works, gives back to OSM, and has a two-way relationship with the community.

I hope this helps

Chris | chris_debian

mnalis, 2023-04-14:
- (ref. Mapilio not being opensource and other issues: https://github.com/kartaview/android/issues/188#issuecomment-1507127472)

mnalis, 2023-04-18:
> "However, we believe that keeping our software proprietary allows us to maintain a level of control over our development process and ensure the quality of our products."

Thanks for forwarding that, @retiolus . As I feared from their initial response, they are not interested in open source community **at all**, but would only prefer that mappers do work for them for free, while they profit without providing anything in return to that community. 

I think they've thus set themselves up to fail - as basically only card with which you can have a chance playing against Facebook is to be open, use FOSS and respect privacy and thus gather all that community that wants open alternative to Facebook/Mapillary. And they've decided to throw away that card. But hey, it is their business, so they should be able to choose how to run (or ruin) it... 

Anyway, nothing interesting for OSM community over there at Mapilio, move along...  Too bad. :disappointed:

chrisdebian, 2023-04-18:
Sadly, I suspect you're right :-(

Chris

chrisdebian, 2024-11-07:
Hi, all.

I've just been adding some information to the OSM Wiki, and specifically to an [existing table](https://wiki.openstreetmap.org/wiki/Street-level_imagery_services#With_photosphere_support), which compares different features of clients such as KartaView. Have a look if you get a moment, and obviously, if you're able to fill in any gaps, please do.

Many thanks,

Chris.

chrisdebian, 2026-10-28:
https://github.com/mapilio/mobile-apps