kartaview/android #184: Releases and Source Code Missing from Repo

chrisdebian, 2022-10-18:
Hi,

I currently have version 5.2.1 (145) installed on my Android device. This version was downloaded from the Google Play Store.

Looking here: https://github.com/kartaview/android/releases

It looks like the last published APK and SRC was published in March 2022. 

Please can you update the repository, so that the source code is available for the latest version of Kartaview. It is my understanding that this is a requirement of the Open Source licence. This will also enable others to fork the code, if they want to review or modify it.

Thank you for your help.

Chris
chris_debian

HolgerJeromin, 2022-10-18:
duplicate of #153

chrisdebian, 2023-02-12:
Duplicate

closed 2023-02-12 by chrisdebian