kartaview/android #187: Delay in processing/ publishing captured imagery?

chrisdebian, 2023-01-01:
Hi, all.

I was a little concerned that Grab may not be processing my captured imagery. Looking at the Profile tab on the Android app, it looks like my most recent imagery was from 7 Dec 22. I know we've had Christmas, but that was 3 weeks ago, and I assume processing is automated. 

Can anyone confirm when imagery was last published, that was uploaded? Is 3 weeks 'normal'?

Many thanks,

Chris
chris_debian

joeybab3, 2023-01-03:
My last track uploaded and processed fine 2022-12-27.
When their server has an issue with a track it stops processing everyone's and it doesn't continue until they do something.
Post your unprocessed track URLs here and tag one of the employees and maybe they will redo it for you

chrisdebian, 2023-01-03:
Hi, Joey. Looks like my recent tracks are now up to date. Maybe they had a backlog.

I'll mark as closed.

Thanks,

Chris

chrisdebian, 2023-01-03:
All good, now!