kartaview/android #188: So how do we make the Android App better?

chrisdebian, 2023-02-13:
Hi, OSM contributors.

Reading some of the issues here on GitHub, it seems like the user community would like the Kartaview Android app to improve its open source credentials. 

Looking forward to the future, I was wondering, as an OSM and imagery contributor, what the app should be able to do. I wanted to take this back to basics and remember what the app was originally intended to do. So thinking back to OpenStreetView and the future, this is my first stab at where the app should be heading.

Please don't delete any of the following, but do add comments, or other things that I haven't yet thought of. Think basics, not feature-creep. We can then fine tune so that Kartaview can decide whether they want to implement the ideas, or whether the community is irrelevant.

1. All code to be free and open source software (FOSS)
2. All code and releases to be made available within one week of any updates. Is target for APK is Google Play and F-Droid?
3. Articulate Key User Requirements (KURs); for example, 'Enable quality of OSM mapping to be enhanced'.
4. Define what the application WON'T do.
5. Define target OS, just Android, or something else?
6. Aim to refactor existing open source code.
7. Hosting of imagery.
8. Hosting for development (CI/CD?).
9. Ability for contributors to export imagery to another app/ repository, if wanting to move to another collection app. This was important for example, when Mapillary was acquired by Facebook (>Meta).
10. Some sort of wiki, which will help contributors to understand the aim of the app and to capture a future requirements 'backlog'. 
11. Reputation: It is important that the app/ collection-client is respected by the user community. 
12. Could the name be improved, so it's aim is more obvious (OpenStreetView2?)
13. ...

Thanks,

Chris

chrisdebian, 2023-02-16:
@ KartaView Devs, do you have any thoughts about the above?

Thanks,

Chris

koalp, 2023-04-03:
> All code and releases to be made available within one week of any updates. Target for APK is Google Play and F-Droid?

If the development is open (with issues on github, gitlab, hosted, ...) , the code would be available immediately (and even before releases. 

> Ability for contributors to export imagery to another app/ repository, if wanting to move to another collection app. This was important for example, when Mapillary was acquired by Facebook (>Meta).

I completely agree, it should be easy to migrate. Shouldn’t the server (and its development) also be FLOSS? I do not find the kartaview server source. Do you know if it is open source ?
If the source isn’t available, I think that there would be two projects to implement from scratch : kartaview-storage and kartaview-web-client.

> Hosting of imagery.

A problem may be that it would take too much disk space (and therefore too much cost) to host. Do you already have ideas for hosting ? And do we have an estimation of disk space needed ?

> Could the name be improved, so it's aim is more obvious (OpenStreetView2?)

The rename from openstreetview to kartaview [seem to have been motivated by an OSMF trademark policy](https://web.archive.org/web/20201210174322/https://blog.improveosm.org/en/2020/11/hello-kartaview/) : “In 2018, the OSM Foundation (OSMF) adopted a Trademark Policy discouraging names like OpenStreet-Thing. This was brought to our attention this year. In support of our good friends, we chose to embrace the name change.”

It would mean that the new name shouldn't contain “OpenStreet”.

chrisdebian, 2023-04-09:
Hi, all.

Just read about Mapilio, based on the open source code of Mapillary. Now to work out a way to move my imagery from KV.

mapilio
[https://mapilio.com/app](https://translate.google.com/website?sl=auto&tl=EN&hl&u=https://mapilio.com/app)
[https://mapilio.com/terms](https://translate.google.com/website?sl=auto&tl=EN&hl&u=https://mapilio.com/terms)
[https://mapilio.com/licenses](https://translate.google.com/website?sl=auto&tl=EN&hl&u=https://mapilio.com/licenses)
It is an app that is based on the open source of Mapillary. It is a project that is starting so it is promising, but it does not have many photos in its repository

Fingers crossed.

Chris

chrisdebian, 2023-04-12:
I'm giving Mapilio a chance, it's maintained and feels modern. Maybe worth asking Mapilio to clarify the open source question.

Thanks,

Chris.

chrisdebian, 2023-04-13:
Hi, all. 

I've had some clarification from the Mapilio Devs.

https://forum.mapilio.com/t/where-is-the-open-source-mapilio-android-app-hosted/112/2

In essence, because the app is still in beta, they are not yet able to publish the code for the Android app. My main concern is that the imagery is covered by the Creative Commons license, and ownership remains with the person who captured the imagery, so this is a good thing.

I'm encouraged by what Mapilio is doing, but I can see why contributors may stick with Mapillary and KV. I'll monitor the situation, and see how it is giving back to the community. For me, I need an app that works, gives back to OSM, and has a two-way relationship with the community.

I hope this helps

Chris | chris_debian

mnalis, 2023-04-13:
> I've had some clarification from the Mapilio Devs.
> https://forum.mapilio.com/t/where-is-the-open-source-mapilio-android-app-hosted/112/2

Thanks for your effort @chrisdebian, it is appreciated. Unfortunately, I cannot say their answer fills me with much confidence we'd see that app open sourced. The same arguments they use (_"it's beta so not yet ready"_, _"we're working on it"_, _"there are already some libs opensourced"_, etc.) have literally being used both by Mapillary and Kartaview too. 

If they want to build confidence in their intentions, they should publish on github source for everything that they legally can, and note what libraries they can't publish yet. I don't care if it the code doesn't work yet, I want to see an open gesture and progress toward that goal. Then we can watch daily commits roll in and see how they advance on removing those non-free dependencies. With no app code and no commits, but only vague promises, it's not convincing at all.

> My main concern is that the imagery is covered by the Creative Commons license, and ownership remains with the person who captured the imagery, so this is a good thing.

Um, yes. But Kartaview images are **also** CC-BY-SA, as are Mapillary images too. So literally no difference regarding that.

Also, [Mapilio ToS](https://mapilio.com/terms) says _"Except for the license you grant below, you do NOT retain all rights in and to your User Content, as between You and Mapilio."_ (emphasis of "NOT" is their, not mine!) 
On the other hand, [kartaview ToS](https://kartaview.org/terms) seem much less aggressive toward contributors.

But neither on them deliver a working open source app, so they don't get to harvest results my time and effort.

> I can see why contributors may stick with Mapillary and KV

It's the bigger userbase which attracts more users. Freedom-wise, all three solutions are about equally unusable.

> For me, I need an app that works, gives back to OSM, and has a two-way relationship with the community.

Absolutely. Before it was acquired by Facebook, I did use their open source command line tools to upload manually taken videos correlated with GPX. But it is poor alternative to an mobile app, and I won't use any app unless it is free enough to be published by f-droid. So, I've stopped contributing that data and use it only locally. 
If some company finds a way that I can easily use opensource app, and that data remains available for others without restrictions, I'll start contributing again. 
IOW, I don't want a relationship with a parasitic company only taking and not giving. As you say, **it has to be two-way relationship. And having at least fully working opensource mobile app is an absolute prerequisite** for that.

chrisdebian, 2023-04-13:
Completely agree @mnalis. Looks like I was overenthusiastic about the new app, I was being unusually optimistic. Back to square one.

Thanks,

Chris.

ozcan-durak, 2023-04-13:
Dear Chris,

I appreciate your post. I just wanted to clarify that while Mapilio has knowledge from Mapillary, it is not based on it. The Mapilio platform and its applications were entirely developed by the Mapilio team.

Best regards,

Coehill, 2024-05-03:
Panoramax is a great option which is already fully open source and on top of that, it is self-hostable and federated!

chrisdebian, 2024-05-03:
> Panoramax is a great option which is already fully open source and on top of that, it is self-hostable and federated!

Wow, that's interesting. So what is the Android equivalent for capturing imagery; the KartaView equivalent, per se?

How do we transfer previously contributed images from KartaView to Panoramas?

Is the imagery easily available to OSM, to improve mapping?

I'm very interested.

Chris

RedAuburn, 2024-05-07:
> Wow, that's interesting. So what is the Android equivalent for capturing imagery; the KartaView equivalent, per se?

The app is still in early stages, but being worked on here: https://github.com/nobelization/panoramax-mobile-app

chrisdebian, 2024-05-07:
Thanks, @RedAuburn Harry. Not quite at the APK stage, yet. Looks promising, though.

Chris

chrisdebian, 2024-12-29:
Also [this](https://play.google.com/store/apps/details?id=com.ravenfeld.panoramax.baba) Panoramax based app 

Chris